IMAP-Science-Operations-Center/imap_processing #37: Move the Contributor Checklist in style guide to PR template?

bourque, 2023-07-14:
@maxinelasp brought up a good suggestion in #19 that perhaps the [Checklist for Contributors of PRs](https://github.com/IMAP-Science-Operations-Center/imap_processing/tree/dev/style_guide#checklist-for-contributors) should exist in a PR template instead of (or in addition to?) the style guide.

What are everyone's preferences?

greglucas, 2023-07-14:
I am torn on checklists in PRs. When I'm a frequent contributor I find them annoying because I know what to do already and sort of just ignore them so they don't really serve a purpose. But, I think it is useful for new contributors to check things off and makes it easier for them to hit all the requirements right away without us needing to comment. My middle ground here would be to put only a few critical checkboxes rather than that long list of many checkboxes.

bourque, 2023-09-12:
We seem to be operating alright without having the checklist in a PR template, so I think we should close this for now.

closed 2023-09-12 by bourque