IMAP-Science-Operations-Center/imap_processing #38: Setup Data processing repository

sdhoyt, 2023-07-14:
**Summary**
A repository should be set up with automated workflows 

**Parent Requirement**
SOC-SDC-L4-62 | Configuration Management | The SDC shall maintain a version controlled Code Management Repository of the data processing software. | Requirement | Necessary to perform science processsing and analysis. | SOC-L3-91 | Inspection
-- | -- | -- | -- | -- | -- | --

**Tasks**
- [ ] Create repository
- [ ] Create automated workflows (check of there's an existing ticket for this

GFMoraga, 2023-07-24:
I make a SPIKE to help explore this.

greglucas, 2023-07-27:
The automated workflows were setup on the initial repository creation: https://github.com/IMAP-Science-Operations-Center/imap_processing/commit/40a266ab735e670f0ede013d0efaa97862f09fb5

Closing this as completed