IMAP-Science-Operations-Center/imap_processing #43: SPIKE - Versioning

bourque, 2023-07-19:
## Description

This spike should explore options for how we version the `imap_processing` software when a release or tag happens. The end goal is to decide on a protocol for how we construct and assign version numbers / IDs to be used in GitHub releases or tags.

## Requirements
- The versioning protocol should associate a version number or ID to a tag or release of the `imap_processing` repository.

### Nice to have or Goal Requirements
- The versioning protocol should make it easy to distinguish which instrument(s) subsystems have changed in a new release or tag (i.e. instrument-specific releases)

## Additional notes

## Existing design
There is no existing design, as we have yet to make any releases or tags.

## Related tickets
- #39

## Follow up tickets
This spike is not considered complete until at least one follow up issue is created. 
- [ ] Placeholder ticket

**Below is the template for the response to this ticket. Add as many solutions as needed, but preferably include 2-5 for discussion. The response should be posted as a comment on this issue, or linked in a comment. **

# Solution 1
Write an overview of the solution here.

## Pros:
- pro 1

## Cons:
- con 1

## Additional notes:
- note

# Solution 2
Write an overview of the solution here.

## Pros:
- pro 1

## Cons:
- con 1

## Additional notes:
- note

# Summary
Write up a summary of your findings, including your preferred solution.

bourque, 2023-07-19:
# Solution 1: Project-wide Semantic Versioning

Semantic Versioning uses a three-part version number of `Major.Minor.Patch` (e.g. `0.1.2`, `0.1.3`, etc). In this solution, it would be applied when any change happens across the repository, regardless of which instrument(s) are affected.

## Pros:
- It is easy to distinguish between major releases (i.e. backwards-incompatible API changes, new major features), minor releases (i.e. new mid- or small-scale features), and bug fixes.
- It is easier to use for dependency management; users or developers can specify version constraints in software environments.
- It is widely used in the field, and so most people should be able to follow our release history and have a rough sense of how we have developed the software.

## Cons:
- It does not make a distinction on which instrument(s) software was changed in a release.
- We additionally need to define what constitutes a major change vs minor change vs bug fix.
- It might lead to very frequent version bumps for minor changes.

## Additional Notes
- While we would not be able to distinguish between instruments in the version number, we could make note of which instrument(s) were affected in release notes.

# Solution 2: Instrument-specific Semantic Versioning 

Use semantic versioning, but keep a separate set of version numbers for each instrument (e.g. `codice-0.1.2`, `codice-0.1.3`, etc.)

## Pros:
- It would make a distinction on which insturment(s) software was changed in a release.
- Someone who is only interested in a specific instrument(s) would only need to be mindful of those instrument's releases

## Cons:
- No project-wide release history
- It would be difficult to make a release for changes that are not instrument-specific (e.g. CI workflows, project documentation, etc.)
- It might not be clear to a user which version to grab if they just want the latest-and-greatest `imap_processing` code

# Solution 3: Project-wide and Instrument-specific Semantic Versioning

Basically the combination of Solutions 1 and 2 -- Have version numbers for both project-wide (e.g. `0.1.2`) and instrument-specific (e.g. `codice-0.1.0`) changes. 

## Pros:
TBD

## Cons:
TBD

# Solution 4: Date-Based Versioning

Date-based versioning uses the date of the release as the version number, in some defined format (e.g. `YYYY.MM.DD`)

## Pros:
- It is easy to see the chronological history of our releases, and tie a particular release to a specific time period

## Cons:
- It does not indicate if the changes are major, minor, or bug fixes
- It is difficult to make multiple releases in the same time period

greglucas, 2023-07-20:
Did you form an opinion on what we should do at all?

My vote for what you describe would be solution 3 because it seems like we should keep track of versions for each instrument as well as the overall pipeline.

For naming, could we make use of namespace packages and put a `__version__` in each of the submodules/packages the instruments will have? I am not sure if we should put instrument names in the version string. Here is a quick example of what I think might work:

```
imap_processing/__init__.py:  __version__ = 1.0.0  # overall project version
imap_processing/swe/__init__.py: __version__ = 2.3.0  # SWEs version number
imap_processing/codice/__init__.py: __version__ = 1.5.0  # CoDICE's version number
```

accessed as:
```python
import imap_processing
from imap_processing import swe

imap_processing.__version__
swe.__version__
```

Another thing that I don't know if we've brought up at all is having `imap_processing` use git submodules. I find them a pain to use most of the time, but this might be an appropriate use-case with subpackages here... But, we can always change over to that later as well I think.

bourque, 2023-07-21:
Thanks @greglucas, I will update the text of solution 3 to reflect what you wrote here. On paper it looks like a great solution, though I haven't really formed any opinions yet.

bourque, 2023-07-21:
@greglucas @tech3371 @sdhoyt @maxinelasp @GFMoraga @laspsandoval @bryan-harter Please share any opinions you have about these versioning schemes, or any pros, cons, or other solutions I may have missed.  Thank you! 🙌🏻

maxinelasp, 2023-07-21:
I definitely would want to see project-wide semantic versioning. Is it super necessary to distinguish by instrument? I think solution 3 is a pretty good solution in that case, but I would want to consider if that is necessary, because it does seem like a fair amount more to track and maintain. If we're putting everything in one repository, I kind of feel like splitting it out by instrument will just introduce confusion.

 I'd prefer to have teams default to updating to the latest version, even if their instrument is not updated, for the following reasons:
1. Security and dependency updates 
2. I wouldn't want one team to get really far behind on the version just because their specific instrument isn't updated - that seems like it could cause problems
3. It will be easier to debug and manage if we just have one version number to check instead of 2 or more.

I'm not particularly against versioning by instrument - I think it has some advantages, and I feel that we will end up doing some versioning for various reasons - but I just feel that at the project level, I'd prefer to just have one version, and I am not seeing any super-compelling reasons to keep them instrument separate for now.

Whatever we decide, +1000 for semantic versioning from me!

bourque, 2023-07-26:
It sounds like solutions 1 and 3 are the leading contenders.  I am mostly indifferent to those, and I think both could work just fine.

That being said, I think I agree with @maxinelasp here -- I also don't really see a compelling reason to also have instrument-specific versions, and I like @maxinelasp's reasons for staying updated. For this reason I lean towards solution 1.

If an instrument team needs to update their specific instrument's code, I don't think it is much more of an ask for them to update the whole package while they are at it, especially if our dev team does our due diligence in quickly distributing our software, making it easy to update, providing good release notes, etc.

bryan-harter, 2023-07-26:
I believe that the original ticket ticket was also about how to relate the software version to the file names of the generated CDF files.  

According to the IMAP file naming convention, the last field of each file will have a versioning number of "vNN-NN", where the first number is the "major" version and the second number is the "minor" version.  I would say that typically on missions, the major version number represents changes to the algorithm, and the minor version represents reprocessing the same file with the same algorithm, but perhaps with updated calibrations or SPICE files.  

So if we start with versioning the repository as a whole, would we increase the version numbers of all the files at once?  For example, if we released imap-processing 2.0.0, would all of the files generated for the mission now begin with "v02-00" in the version field?  Or would we keep some sort of mapping that says "imap-processing 2.0.0 corresponds to v01-00 for mag, v02-00 for swe, v03-00 for IDEX, etc.  

Sorry if I'm retreading an old conversation here!

bourque, 2023-07-26:
Thanks @bryan-harter, this is quite helpful.  It sounds to me like the versioning schema we use for data files could (or should) be slightly different to what we use for code.  Which I think is fine, as long as we document what we are doing in the style guide.

Based on the votes I saw in Slack and the discussion here, it sounds like we should go with Solution 1 for versioning the `imap_processing` repository.  As for how that translates to the version used in the filename conventions, we could decide on a mapping scheme like you mentioned.  I think I will open a separate issue for that question, since it is slightly different from "how do we version the project-wide code". (Or maybe I need to look through the slack/github history for that convo if it was already had)

bourque, 2023-07-28:
Based on the discussion here and the votes received in the slack convo about this, we are going to so with Solution 1.  If we find that our lives would be better with instrument-specific versioning in addition to this, we can implement that sometime later.

closed 2023-07-28 by bourque