IMAP-Science-Operations-Center/imap_processing #44: SPIKE - Configuration Management

GFMoraga, 2023-07-24:
## Description

This SPIKE should explore what needs to go into a version controlled Code Management Repository of the data processing software. This is necessary to perform science processing and analysis.

## Requirements
- Create automated workflows (GitHub actions)

### Nice to have or Goal Requirements
-  Nice to have the parameters that go into this automated workflow

## Additional notes
Should this be a completely different repo (called Code Management)  OR should this be done in imap_prosessing? 

## Related tickets
- `#38` 
- `#39` 

## Follow up tickets
This spike is not considered complete until at least one follow up issue is created. 
- [ ] Placeholder ticket

GFMoraga, 2023-07-24:
# Solution 1
Custom workflows

## Pros:
- Modularity and Reusability
- Automate Complex Processes
- Tailored to Project Requirements

## Cons:
- Complexity and Learning Curve: Creating custom workflows often requires knowledge of YAML syntax and GitHub Actions syntax.
- Maintenance Overhead: While custom workflows offer flexibility, they can also require more maintenance effort.
- Time Investment: Designing and testing custom workflows can take time, which may initially slow down the development process.

## Additional notes:
- This could benefit IMAP due to the complexity of the project

# Solution 2
Continuous Integration (CI)

## Pros:
- Early Detection of Bugs
- Faster Feedback Loop
- Streamlined Merge Process

## Cons:
- Setup Complexity
- Build Time Overhead: Running tests and other tasks during CI can increase build time, especially for larger projects, leading to longer feedback cycles.
- Integration Challenges: Integrating CI with legacy systems or complex architectures may require additional effort.

## Additional notes:
- This is the most popular automated workflow

# Summary
We need to figure out what workflows benefit IMAP as a whole.

GFMoraga, 2023-07-24:
@bourque @greglucas @sdhoyt @tech3371 @maxinelasp what do you think of this SPIKE? Feedback and input would be great!

maxinelasp, 2023-07-24:
Can you say a little more about which option you think is better, and why? I'd also like to see more of a summary of what the solution actually is. Continuous integration is more of a style of deployment than a particular piece of technology, so I'm not sure if the comparison makes sense. 

So, additional details I'd like to see in each solution:
- description of exactly what the solution is
- in additional notes, it would be good to see what the implementation of each solution would look like - so, for CI, would that involve continuous deployment as well? Would CI run for the whole project each time, or be modular? What do we want both options to provide? For example, I'd like to have any option automatically run tests. 

Just generally, I'm not sure what you're picturing when you're talking about custom workflows or CI. If you aren't sure what things are important to the project or what we would want to see, then you can add questions to the end for discussion.

greglucas, 2023-07-24:
Looking at the original issue, I think your original questions were good: https://github.com/IMAP-Science-Operations-Center/imap_processing/issues/38#issuecomment-1648176138

> There is already a test.yml in the imap_processing/.github/workflows.
Is this new repo outside of imap_processing?
If not, should I just be updating the imap_processing/.github/workflows?

My take on this is that you nailed it and we already have `test.yml` which is our Continuous Integration for the processing workflow. So, I think the original issue may already be completed and we just have to document that, but it is probably good to wait for @sdhoyt to weigh in on that original issue before going too far down all of these investigations.

I also agree with everything that @maxinelasp wrote above.

sdhoyt, 2023-07-27:
Yeah, I think this is done. This was to setup the IT infrastructure for configuration management and tools to maintain it. So I think once we have the repository setup and the test workflow setup to help us maintain it, we can call this done.

closed 2023-07-27 by greglucas