IMAP-Science-Operations-Center/imap_processing #52: Enable doc builds for PRs

greglucas, 2023-08-05:
### Topic:

It would be nice to enable doc builds on PRs, so that people can automatically see what their documentation will look like. I think this just takes an admin to enable and get the webhooks going:
https://docs.readthedocs.io/en/stable/guides/pull-requests.html

cc @bourque and @maxinelasp as two people interested in the docs.

bourque, 2023-08-06:
I tried to do this but it appears that I don't have admin privileges for the project on RTD.

maxinelasp, 2023-08-07:
I think @sdhoyt probably is the one with admin privileges for imap_processing, and I have them for sds-data-manager. I can go and add the team members as admins today so we can all have access to that.

sdhoyt, 2023-10-17:
This issue has been resolved. Docs are building as part of Github Actions in PRs

closed 2023-10-17 by sdhoyt