IMAP-Science-Operations-Center/imap_processing #67: Create a template file for XTCE Packets

maxinelasp, 2023-08-21:
### Topic:
Create a template with common fields and parameters for all (or most) XTCE packet files. 
### Description:
This file should include things like the CCSDS header, which are common to all files. The goal is to have one template which we can copy out of for other XTCE files.

### Requirements
Include some kind of comment making it clear that it came from a template and where you can find the template.

maxinelasp, 2023-08-21:
May be related to #6 for an XCTE script builder - @GFMoraga What do you think? Does it make sense to integrate some kind of template into your script?

GFMoraga, 2023-08-21:
This is related! I think the Python code I have will help with this. The is to format the code with '''information''' so the user can understand/change inputs when needed. 

'''Create CCSDS Header parameters, which should be the same for all IMAP instruments'''

'ccsds_parameters = [
    {"name": "VERSION", "parameterTypeRef": "uint3", "description": "CCSDS Packet Version Number (always 0)"},
    {"name": "TYPE", "parameterTypeRef": "uint1", "description": "CCSDS Packet Type Indicator (0=telemetry)"},
    {"name": "SEC_HDR_FLG", "parameterTypeRef": "uint1", "description": "CCSDS Packet Secondary Header Flag (always 1)"},
    {"name": "PKT_APID", "parameterTypeRef": "uint11", "description": "CCSDS Packet Application Process ID"},
    {"name": "SEG_FLGS", "parameterTypeRef": "uint2", "description": "CCSDS Packet Grouping Flags (3=not part of group)"},
    {"name": "SRC_SEQ_CTR", "parameterTypeRef": "uint14", "description": "CCSDS Packet Sequence Count (increments with each new packet)"},
    {"name": "PKT_LEN", "parameterTypeRef": "uint16", "description": "CCSDS Packet Length (number of bytes after Packet length minus 1)"},
    {"name": "SHCOARSE", "parameterTypeRef": "uint32", "description": "CCSDS Packet Time Stamp (coarse time)"}]'

GFMoraga, 2023-08-21:
Or are you suggesting something like:

### CCSDS headers

<xtce:ParameterSet>
			<xtce:Parameter name="VERSION" parameterTypeRef="uint3">
				<xtce:LongDescription>CCSDS Packet Version Number (always 0)</xtce:LongDescription>
			</xtce:Parameter>
			<xtce:Parameter name="TYPE" parameterTypeRef="uint1">
				<xtce:LongDescription>CCSDS Packet Type Indicator (0=telemetry)</xtce:LongDescription>
			</xtce:Parameter>
			<xtce:Parameter name="SEC_HDR_FLG" parameterTypeRef="uint1">
				<xtce:LongDescription>CCSDS Packet Secondary Header Flag (always 1)</xtce:LongDescription>
			</xtce:Parameter>
			<xtce:Parameter name="PKT_APID" parameterTypeRef="uint11">
				<xtce:LongDescription>CCSDS Packet Application Process ID</xtce:LongDescription>
			</xtce:Parameter>
			<xtce:Parameter name="SEG_FLGS" parameterTypeRef="uint2">
				<xtce:LongDescription>CCSDS Packet Grouping Flags (3=not part of group)</xtce:LongDescription>
			</xtce:Parameter>
			<xtce:Parameter name="SRC_SEQ_CTR" parameterTypeRef="uint14">
				<xtce:LongDescription>CCSDS Packet Sequence Count (increments with each new packet)</xtce:LongDescription>
			</xtce:Parameter>
			<xtce:Parameter name="PKT_LEN" parameterTypeRef="uint16">
				<xtce:LongDescription>CCSDS Packet Length (number of bytes after Packet length minus 1)</xtce:LongDescription>
			</xtce:Parameter>

### SequenceContainer

<xtce:SequenceContainer name="CCSDSPacket">
				<xtce:EntryList>
					<xtce:ParameterRefEntry parameterRef="VERSION" />
					<xtce:ParameterRefEntry parameterRef="TYPE" />
					<xtce:ParameterRefEntry parameterRef="SEC_HDR_FLG" />
					<xtce:ParameterRefEntry parameterRef="PKT_APID" />
					<xtce:ParameterRefEntry parameterRef="SEG_FLGS" />
					<xtce:ParameterRefEntry parameterRef="SRC_SEQ_CTR" />
					<xtce:ParameterRefEntry parameterRef="PKT_LEN" />
					<xtce:ParameterRefEntry parameterRef="SHCOARSE" />
				</xtce:EntryList>
			</xtce:SequenceContainer>

maxinelasp, 2023-08-21:
Either is fine, depending on how your script works. But, it's my preference that it's in a different file, so it's easy to update and to see when it's getting updated.

GFMoraga, 2023-08-21:
Gotcha. Something that is named "xtce_packet_temp.md/yml/py/xml" ?

maxinelasp, 2023-08-22:
Yes, except write out "template", because I read that as "temporary"! But yes, basically some kind of template which can be shared across different instruments. But, don't make it markdown, because that will be annoying for programming. If you make it something non-xml (which I think would make it easier to edit) make sure you add something in your script to output it as XML. That way, someone who is using the template but not your script can still use the template.

GFMoraga, 2023-08-22:
@maxinelasp I just saw this, and I think I could add this to my current PR, but can do a new one too. Could you look at the open PR and see what you like in the sense of making a template? I tried to add guidelines in the script, so I would take from those to make this template.

maxinelasp, 2023-08-22:
Either adding it to the current PR or adding to a later one would be fine. It looks like what you have is good in the PR, you would just need to pull out some of the template generation into a new "generate_ccsds_header_xml" type function, and then it is easy to import that method and call it to get the XML. I will take a look at that PR some time this week, ping me if I haven't by like Thursday or Friday.

tech3371, 2023-08-23:
I am checking in to see if I need to create XTCE template file (which I don't mind doing) or what @GFMoraga has done this?

GFMoraga, 2023-08-23:
@tech3371 look at my PR and see. I was going to make a CCSDS header template like Maxine suggested.

maxinelasp, 2023-08-23:
I think what you have is great, Gabriel! Maybe Tenzin can look through my suggestions and see if that PR completes this ticket, or if there's some additional work to be done there. It might be worth just merging that PR quickly and then coming back to it to rework it to be more of a general template. But I think what you have for the template and the code generation tools in that PR looks good! (PR: #72 )

tech3371, 2023-08-23:
If we can generalized his work on current PR #72 , Then what he has done is far better than what I would have done, which is just XTCE file with same example and more comments.

GFMoraga, 2023-08-23:
So, I add a new file to #72 which is the 'generate_ccsds_header_xml.py' 

How do we close this issue and do y'all @maxinelasp @tech3371 think this is a good start? We can move the file too.

maxinelasp, 2023-08-23:
Yes, I think that's definitely sufficient for this issue!

closed 2023-08-23 by maxinelasp