IMAP-Science-Operations-Center/imap_processing #75: Xtce build follow up: make updates on closed PR #72

GFMoraga, 2023-08-23:
### Summary

    - Making the tests more in the style of unit tests, and maybe adding more of them if there are more areas we can test
    - Moving docstrings around to be compliant with PEP257/numpydocs
    - Adding said documentation to the imap_processing docs (as well as any other outward-facing docs we want to add)
    - Utilizing classes, config files, or argument parsing to abstract out parameters from the code
    - Make the code more generalized, generally speaking

### Requirements

The more specific the better.
  - Review closed PR #72 
  

### Related tickets

#72

GFMoraga, 2023-08-23:
@bourque I copied your comment from the PR as a starter. I will go back and edit/update.

bourque, 2023-08-23:
Thanks @GFMoraga. I am also happy to help out with some of this!

GFMoraga, 2023-08-24:
@tech3371 how do I do this? "We need to add pandas, openpyxl to poetry dependency list" in poety.toml?

maxinelasp, 2023-08-24:
You can also look at `space_packet_parser` for [examples](https://github.com/medley56/space_packet_parser/blob/master/space_packet_parser/parser.py) of class names/structure.

maxinelasp, 2023-08-24:
Also, per your poetry question, I will write up a document about using poetry which we can use as a start.

tech3371, 2023-08-25:
@GFMoraga I found this book very helpful. I read first few chapters. Chapter 2 talks about how to name functions well. I hope it helps.
https://github.com/jnguyen095/clean-code/blob/master/Clean.Code.A.Handbook.of.Agile.Software.Craftsmanship.pdf

closed 2023-09-27 by GFMoraga