IMAP-Science-Operations-Center/imap_processing #81: Create a document covering how to git rebase

maxinelasp, 2023-08-28:
### Overview

To keep our git history clean, we request that contributors run `git rebase` and squash unneeded or repetitive commits. This is not as easy as it sounds, so we should write up a guide for how to run this command for those who are not git wizards. 

### Related documents or information sources

https://git-scm.com/book/en/v2/Git-Branching-Rebasing
https://stackoverflow.com/questions/40009820/git-rebase-one-branch-on-top-of-another-branch
https://www.atlassian.com/git/tutorials/rewriting-history/git-rebase

greglucas, 2023-08-28:
> so we should write up a guide for how to run this command for those who are not git wizards

Instead of writing our own guide, what about just adding those links you have there into the documentation with a few sentences describing why we are including the links? If we want to keep it in the scientific python community, we could include a link to Numpy's development page. (I think Scientific Python is working on some kind of git-help guide for the community to reference but I couldn't find it yet)

https://numpy.org/doc/stable/dev/development_workflow.html#rebasing-on-main

maxinelasp, 2023-08-28:
The existing documentation is a little complicated in my opinion - but that document you linked is better than anything I've been able to find. I was picturing a very simple, "here's the most basic use case," with a few links out to other useful information.

greglucas, 2023-08-28:
> The existing documentation is a little complicated in my opinion
I agree!

A simple example with a few lines of code sounds good. I was just worried that we were also going to add in the how to recover from mistakes to the guide as well. Rebasing is definitely not trivial, so adding the simple case and then linking out for additional help is a great approach 👍

closed 2025-05-02 by greglucas