IMAP-Science-Operations-Center/imap_processing #124: [L5] Produce L1A CDF formatted data products for IMAP-Hi

tech3371, 2023-09-06:
### Summary of the L5 requirement

The SDC needs to produce L1A CDF formatted data products for the IMAP-Hi instrument. The data products for L1A can be found in the IMAP-Lo algorithm document here: https://lasp.colorado.edu/galaxy/display/IMAP/IMAP+Algorithm+Document+from+Instrument+Teams

### Parent requirement

SOC-SDC-L4-29 | SDC Execution of L1 Processing Software | The SDC shall be capable of executing L1 processing software to produce L1 data products within seven (7) days of receipt of all requisite data inputs. | Requirement | The SDC produces L1 data products via Instrument Team supplied algorithms and SDC produced processing code. | SOC-L3-57 | Test | SIT-3: L0 -> L1 processing | SIT-3
-- | -- | -- | -- | -- | -- | -- | -- | --

### Tasks

**Tasks**
- [ ] Create XTCE for housekeeping, direct event and histogram packets
- [ ] Unpack and process those packets to L1A
- [ ] Write data to CDF file
- [ ] Validate data

subagonsouth, 2025-05-13:
@tech3371 @sdhoyt , Are these tickets with the `Req: L5` labels being used to track anything? These are from before I was on the project and I don't know if these are something that I can cleanup/close or not.

sdhoyt, 2025-05-14:
@subagonsouth No, all the L5 issues can probably be deleted unless you were specifically using it. This was my failed attempt to decompose the level 4 requirements into more testable requirements and track them in Github and we've moved away from doing this. Sounds like you're not using this, so I'll close it with this comment, but feel free to reopen if you are.

closed 2025-05-14 by sdhoyt