IMAP-Science-Operations-Center/imap_processing #135: [L5] SDC should release L1 processing production code

sdhoyt, 2023-09-07:
### Summary of the L5 requirement

The SDC should deliver releases of L1 processing production code for all instruments. This can be releases to pypi or a similar platform.

### Parent requirement

  | SOC-SDC-L4-66 | Produce Production Science Processing Code | The SDC shall produce production code for all instrument teams. | Requirement | Necessary to perform science processsing and analysis. | SOC-L3-57SOC-L3-91 | Test | SIT-3: L0 -> L1 processingSIT-4: L1 -> L2 processing | SIT-4
-- | -- | -- | -- | -- | -- | -- | -- | -- | --

### Tasks

- [ ]

bourque, 2023-10-04:
I think this might be completed now that we have a GitHub Actions workflow that will automatically upload the package to PyPI upon release (introduced in #180)

greglucas, 2023-10-06:
> I think this might be completed now that we have a GitHub Actions workflow that will automatically upload the package to PyPI upon release (introduced in https://github.com/IMAP-Science-Operations-Center/imap_processing/pull/180)

I agree, I think we should link out to those runs demonstrating that this happened for a release.

Are we closing issues before L1 is complete for all instruments? Or are we waiting to close this until we do a release with everything contained in it?

closed 2025-05-05 by bourque