IMAP-Science-Operations-Center/imap_processing #335: Ideas for updates to CDF documentation

bourque, 2024-02-07:
- Remove filename conventions section, instead link to existing naming conventions docs in style guide
- Add *example* of global attributes that will be common across all files (we currently describe the contents in great detail, but don't provide a final example of what it looks like in a CDF file)

maxinelasp, 2024-02-07:
- In API docs, explain how to use imap-data-access for API access, or make it very clear that that is the recommended way to access the API in most cases
- In the CDF docs, link out to variable names defined by SPDF (for example, for quality flags)

bourque, 2024-03-21:
#368 takes care of the first bullet

bourque, 2024-06-04:
I've created separate issues for these, so I think this issue can be closed.

closed 2024-06-04 by bourque