IMAP-Science-Operations-Center/imap_processing #378: Set up Zenodo linking and citation information

greglucas, 2024-03-28:
### Summary

We should add a citation entry to the repository, and start deploying to Zenodo so people can cite the releases.
https://citation-file-format.github.io/
https://docs.github.com/en/repositories/archiving-a-github-repository/referencing-and-citing-content

### Requirements

  - [ ] Link Zenodo to the `imap_processing` repository
  - [ ] Add a citation.cff file with the proper metadata to the `imap_processing` repository

vmartinez-cu, 2024-03-28:
Does LASP have a Zenodo account?

greglucas, 2024-03-28:
That is a great question, I don't know. I have always authorized things through my GitHub-linked accounts. Are there organizational ways to handle it to get better DOIs and things? Might be worth asking in the slack channels for open-science/data-management.

vmartinez-cu, 2024-03-28:
My experience minting DOIs has been for datasets and that involved an organization level account with DataCite, which wasn't free. DOIs for software is new to me, and a zenodo organization level account doesn't seem necessary for this, based on information I've gathered so far. We can link to the LASP community in zenodo to keep cited software organized at an organization level. 

I assume we don't have an initial release for this code just yet? I'm looking into what stage of the development process is a good time to create the DOI since they're linked to software versions. I'm not clear if we can/should create a DOI before version 1 is completed. I'll try to find out.

For now, Zenodo has been linked to the repo! Will look into citation.cff next

greglucas, 2024-03-28:
Awesome!

> I assume we don't have an initial release for this code just yet?

We do: https://github.com/IMAP-Science-Operations-Center/imap_processing/releases

> For now, Zenodo has been linked to the repo!

Fantastic, I _think_ that will automatically pick up any releases we make now 🎉  so we can check off the first box!

vmartinez-cu, 2024-03-28:
Scratch that. I linked it to my forked repo...The parent repo isn't listed in my zenodo account for some reason. I'm a member of the IMAP-Science-Operations-Center repository. Does this automatically make me a member of the imap_processing repository? Just wondering if it could be a membership related issue. Though IMAP-Science-Operations-Center also doesn't show up in my list of repositories in Zenodo. I'll look into the issue

greglucas, 2024-03-29:
I didn't see it either, and then I had to click the "sync now" button and then it showed up.
[image]
I just enabled it under the organization, so I think we should be good.

> I'm a member of the IMAP-Science-Operations-Center organization.

Do you see it now after I checked the box in your own listing? I wonder if it has to do with permissions and you being a "member" and not an "owner" so an owner is the only one who can authorize access possibly.

closed 2024-04-05 by vmartinez-cu