IMAP-Science-Operations-Center/imap_processing #689: SPICE - Update met_to_j2000ns

laspsandoval, 2024-07-08:
### Summary

Update /Users/lasa6858/imap_processing/imap_processing/cdf/utils/met_to_j2000ns so that it is more accurate by using the spacecraft clock kernel.

MET - spacecraft ticks (close to a second)
SCLK - shows how to adjust clock ticks to a true time

kernel location: https://lasp.colorado.edu/galaxy/display/IMAP/Data

You will need to 
1. furnish the sclk kernel similar to how it works in spice_examples.py:
with spice.KernelPool(kernels): etc.
2. Use relevant tool/tools listed here to update the met: https://spiceypy.readthedocs.io/en/main/documentation.html#spiceypy.spiceypy

### Requirements

The more specific the better.
  - Implement SPICE SCLK in met_to_j2000ns

subagonsouth, 2024-07-09:
For this ticket I suggest that the function should ignore the need to furnish the kernels. Assume that the needed kernels will already be furnished. That piece will eventually be taken care of by the decorator that I will implement. For testing, you will need to furnish the kernels noted in the ticket description in the test function before calling `met_to_j2000ns`.

subagonsouth, 2024-07-09:
I would also suggest that `met_to_j2000ns` be a wrapper around a function called `met_to_j2000` which returns double precision floating point seconds since J2000. That function in turn should essentially vectorize the underlying spiceypy function that does the conversion.

vmartinez-cu, 2024-07-10:
I still have some reading up to do on SPICE. What is meant by "furnish" kernels?

subagonsouth, 2024-07-10:
> What is meant by "furnish" kernels?

When you furnish a kernel, by calling `furnish(path_to_kernel)`, you are really just "loading the kernel into the program". This lets SPICE know about the kernel. It opens the kernel and reads some metadata to understand what information the kernel provides. That is really all furnishing does.

This page has a nice overview of kernels: https://naif.jpl.nasa.gov/pub/naif/toolkit_docs/C/info/intrdctn.html

subagonsouth, 2024-07-10:
One piece of background info that I thought might not be obvious is that SPICE source code is written in FORTRAN. NASA's Navigation and Ancillary Information Facility (NAIF) created a C wrapper for the FORTRAN code and provides compiled shared objects that can be called from C. The `spiceypy` package essentially just wraps the C code such that it can be called from python. If you look at the `spiceypy` documentation, all of the function documentation link directly to the NAIF CSPICE documentation.

PR #818 (2024-09-10): 689 spice update met to j2000ns
# Change Summary

## Overview
Moved the `met_to_j2000ns` function from the cdf/utils.py to spice/time.py and re-implemented that function to use SPICE tools for the time conversion.

## Updated Files
There are lots of updated files as a result of moving the function `met_to_j2000ns`. I won't list all of those here. The most important changes are in the listed files below:
- imap_processing/cdf/utils.py
   - moved met_to_j2000ns out of this file
- imap_processing/spice/__init__.py
   - added global constant declaration of IMAP_SC_ID
- imap_processing/spice/time.p [... cut]
closed 2024-09-10 by subagonsouth