IMAP-Science-Operations-Center/imap_processing #710: BUG - mypy should only be run over our actual code

greglucas, 2024-07-24:
### Description of the issue

Currently, we run mypy in the current working directory: https://github.com/IMAP-Science-Operations-Center/imap_processing/blob/bf31f33cdee695537a21ee3d89ecd88a9c4d2241/.pre-commit-config.yaml#L45-L49

If someone adds an extra file without type hints like `temp.py` as a testing script, or puts their virtual environment in the current directory then mypy will try to run over those files as well.

### Steps to reproduce the issue

Create a temporary file or virtual environment without type hints in your local directory, and run `mypy .` and it will fail.

### Expected behavior (What should happen)

Only run over the project directory `imap_processing/`.

### Actual behavior (What does happen)

```
venv3.12/bin/jp.py:12: error: Function is missing a return type annotation [no-untyped-def]
venv3.12/bin/rst2odt_prepstyles.py:29: error: Function is missing a type annotation [no-untyped-def]
```

### Code Snippet:

```python
Code
```

### Additional notes, affected areas, and suggested fixes

Changing mypy to run only over `imap_processing/`. I also suggest adding the current mypy arguments from the pre-commit hook into the `pyproject.toml` configuration file instead. That way anyone could run `mypy` and get the proper ignores/additions without needing to add command line arguments.

daralynnrhode, 2024-07-24:
@greglucas  From my understanding, the toml and precommit set up is strange. From what I can find, I don't think it is possible to disable the errors listed in the pre-commit in the toml file. The errors that are allowed to be disabled in the toml are broader more general errors that could include the specific errors in pre-commit, but they would also include other errors that we would want to check. 
I do believe all of these can disabled from the command line, though, I'm still messing around with it.

greglucas, 2024-07-24:
It may not be a one-to-one mapping so you might not be able to move all of them (maybe some of them aren't needed anymore even?). But a few of them I think could be moved over.

`--explicit-package-bases`: https://mypy.readthedocs.io/en/stable/config_file.html#confval-explicit_package_bases
https://mypy.readthedocs.io/en/stable/config_file.html#example-pyproject-toml

daralynnrhode, 2024-07-24:
I think I have the errors figured out and excluded from both call types, `pre-commit run --all` and `mypy .` but each run produces different errors that in no way overlap.  So I still have to figure out where they pyproject and config and diverging. 

@greglucas Just to clarify, when you say you want it to run over the imap_processing directory, do you mean the repo folder `imap_processing` or `imap_processing/imap_processing` with would exclude the tools folder as well as the docs?

greglucas, 2024-07-24:
> @greglucas Just to clarify, when you say you want it to run over the imap_processing directory, do you mean the repo folder imap_processing or imap_processing/imap_processing with would exclude the tools folder as well as the docs?

Ahh, yes, good call on other locations if we want to check those, mostly I was getting at "it would be nice if it wasn't a blanket `.` check".

I think you could use `file = ['imap_processing', 'docs', 'tools']` or for whatever directories you also wanted to include.

daralynnrhode, 2024-07-24:
I assume you are talking about the instance where pre-commits are ran and not just the individual mypy check?

greglucas, 2024-07-25:
> I assume you are talking about the instance where pre-commits are ran and not just the individual mypy check?

I'm not sure I totally follow the question, so let me know if I'm not addressing the question. I would expect the pre-commit check to run the command `mypy` without any extra arguments if possible, that way a user would only have to run `mypy` locally and get the same settings that pre-commit is using (but without having to know the extra command line arguments).

daralynnrhode, 2024-07-25:
Ah yes! So I am confident that I can configure the pre-commit hook to run over only the desired directories mentioned above without any extra commands. This would be done in the args section of the hook.
Currently, I don't think there is a way to apply this to when a user _only_ runs a mypy check. Mypy when ran by itself has to have module to check passed in with it. So `mypy [file or directory to check]` instead of just `mypy`. From what I have figured out, command line arguments take precedence over any configuration file. So even if it is stated in the `pyproject.toml` to only check certain folders, the command line arg would take precedence. 
Does that make sense? The mypy documentation is a little hard to digest so I may also just be missing something but from my current research that is what I have concluded.

PR #731 (2024-08-09): Mypy Update
# 

## Overview

Kinda long, so sorry for that. Feel free to only check the files that concern your instrument.

Ensuring that both pre-commit mypy and a local mypy check will return the same outputs. In doing so, more errors were found and will be fixed within this PR.

numpy is weird:
As a way to avoid redeclaring a type, ie. `indices: np.ndarray = np.where(valid)[0]` that should be inferred to be a `np.ndarray` because of what the `np.where()` returns, but is not. Instead function returns of `np.ndarray` were changed to `npt.NDArray` from the numpy typing library. I believe in m [... cut]
closed 2024-08-09 by daralynnrhode