IMAP-Science-Operations-Center/imap_processing #717: BUG - logger warnings when only importing a module

greglucas, 2024-07-26:
### Description of the issue

There are extra warnings in the log stream when import `imap_processing.utils` that can't be avoided.

```
WARNING:root:Required schema 'RESOLUTION' attribute not present for epoch
WARNING:root:Required schema 'DISPLAY_TYPE' attribute not present for epoch
```

### Steps to reproduce the issue

```
import imap_processing.utils
```

### Expected behavior (What should happen)

No warnings

### Actual behavior (What does happen)

logging to stdout stream
```
WARNING:root:Required schema 'RESOLUTION' attribute not present for epoch
WARNING:root:Required schema 'DISPLAY_TYPE' attribute not present for epoch
```

### Code Snippet:

```python
Code
```

subagonsouth, 2024-07-26:
Related to this, I think we need to have different sets of required attributes for the different variable types such as `data`, `support_data`, `metadata`, `time`. Documentation here: https://spdf.gsfc.nasa.gov/istp_guide/vattributes.html

greglucas, 2025-01-31:
The import doesn't warn anymore. I assume something was fixed by the upgrade to SAMMI. @subagonsouth, is your comment: https://github.com/IMAP-Science-Operations-Center/imap_processing/issues/717#issuecomment-2253027516 still relevant or did SAMMI add the ability to do that as well? Lets open a separate issue if that is still open.

closed 2025-01-31 by greglucas