IMAP-Science-Operations-Center/imap_processing #719: BUG - cdf_manager.add_global_attribute() is not working as expected

tech3371, 2024-07-26:
### Description of the issue

cdf_manager.add_global_attribute() was adding additional variables to global attrs but was not return as expected. This is due to get_global_attributes() filtering out added variables when that function is called. I am not sure of immediate fix for this.

### Expected behavior (What should happen)

should return additional global attrs that was added by user that's outside of required CDF attrs.

### Code Snippet:

```python
Code
cdf_manager = ImapCdfAttributes()
cdf_manager.add_instrument_global_attrs("swapi")
cdf_manager.add_global_attribute("Data_version", data_version)
cdf_manager.add_global_attribute(
        "sweep_table", f"{sci_dataset['sweep_table'].data[0]}"
    )
```

### Additional notes, affected areas, and suggested fixes

It works with `Data_version` because it's one of the required attrs but it doesn't work with `sweep_table` because it gets filter out by get_global_attributes()

greglucas, 2025-01-31:
@tech3371, is this still an issue after we've switched over to SAMMI now?

tech3371, 2025-02-03:
I haven't had chance to check it because I have moved that parameter into data variable since things changed.

greglucas, 2025-02-03:
OK, can we close this then until we get a new reproducer / issue?

tech3371, 2025-02-03:
sounds good!

closed 2025-02-03 by tech3371