IMAP-Science-Operations-Center/imap_processing #724: Create website page for job tracking and status

bourque, 2024-07-30:
@greglucas mentioned to me that it would be good to have a page on the SDC website that is dedicated to job tracking and status. It would be great if a user could click on a job ID and be able to get the logs for the job from AWS.

This would likely require two new API endpoints, e.g. `GET job-statuses` and `GET job-log` (a separate issue in `sds-data-manager` will be created for these).

bourque, 2025-03-19:
Also assigning @tech3371 here who has some time to work on this

bourque, 2025-06-11:
This is completed now (See https://github.com/IMAP-Science-Operations-Center/sds-data-manager/issues/316)

greglucas, 2025-06-11:
Is it? I don't see anything on the dev website: https://dev.imap-mission.com/ and I thought that Tenzin hadn't finished setting up the job log endpoints. I put this down as high-priority things in the last stand-up we had on Monday because it would have really helped us to have this for the SIT development I thought. If we have this already, we missed an opportunity to use it.

bourque, 2025-08-29:
This was completed ahead of SIT-5!

closed 2025-08-29 by bourque