IMAP-Science-Operations-Center/imap_processing #845: codecov check should not include test files

subagonsouth, 2024-09-16:
### Algorithm Description:
Our code coverage metrics shouldn't care about test directories.

greglucas, 2024-09-16:
I disagree with this. I believe that we want all tests to be run and know if tests are not being run. If codecov doesn't see a test being run, that means the test is pointless. We should have at least one runner to run every test if we can and get full coverage in the tests/ folder.

I think others have this same thought as you though, so I may be the only one who cares about test coverage 🤷 and we should actually do this.

greglucas, 2024-09-16:
Consensus on the call was to ignore `conftest.py` fixture files, but leave the test coverage in tact.

closed 2024-09-30 by tech3371