IMAP-Science-Operations-Center/imap_processing #897: Integrate SPICE into GLOWS L1B

maxinelasp, 2024-10-01:
# Summary
Glows L1B requires some inputs from SPICE, including spacecraft location information. Once SPICE tools are completed, integrate them into GLOWS L1B. 

Depends on SPICE tooling, so blocked for now.

subagonsouth, 2024-10-01:
The notes I took during our SPICE meeting indicated GLOWS needs the following list (checked items are available)
GLOWS Needs:
a. IMAP spacecraft	b.
 - [ ] Spin axis orientation
 - [x] Spin phase
 - [ ] Spin phase change rate
 - [x] Spacecraft position and velocity and their uncertainty
    - No uncertainty in SPK I don't know if uncertainty is possible - Tim
 - [ ] IMAP x, y, z axes and rates of change of those vectors (ECLIPJ2000)
 - [x] IMAP state vectors (position and velocity Relative to Sun, Earth)
 
b. GLOWS instrument
 - [ ] GLOWS spin phase
 - [ ] Instrument pointing at given time
    - Cartesian heliocentric coordinates in ecliptic reference system???

maxinelasp, 2025-07-10:
Partly completed in branch glows_l1b_spice

PR #2026 (2025-08-19): GLOWS: Adding SPICE parameters to L1B
# Change Summary
closes #897 

## Overview

Added and updated SPICE related test in GLOWS L1B code.

## Updated Files

- imap_processing/glows/l1b/glows_l1b.py
   - add new variable and it's dimension
- imap_processing/glows/l1b/glows_l1b_data.py
   - main code that updates SPICE related stuff

## Testing

- imap_processing/tests/conftest.py
- imap_processing/tests/glows/conftest.py
- imap_processing/tests/glows/test_glows_l1b.py
- imap_processing/tests/glows/test_glows_l1b_data.py