IMAP-Science-Operations-Center/imap_processing #994: HEALPix Utils function

tech3371, 2024-10-08:
some links:
https://healpy.readthedocs.io/en/latest/install.html
https://healpix.sourceforge.io/

subagonsouth, 2024-10-09:
It is not clear to me that we actually need to implement this. If the L2 code was modularized to the extent that any binning routine could be plugged in, it would allow whoever develops the Mapper tool to just plug in a binning function that would bin to HEALPix rather than Lat/Lon grid. It my be moot since Ultra is now requesting HEALPix number be included in their pset product.

nkerman, 2025-05-05:
Yes, I think it's fair to close this, at least from the Ultra Standpoint. @subagonsouth do you have any objection to closing this?

subagonsouth, 2025-05-06:
Agree. Closing as duplicate.

closed 2025-05-14 by nkerman