IMAP-Science-Operations-Center/imap_processing #1296: One time conversion function to rule them all

subagonsouth, 2025-01-24:
It seems like we have several functions with slightly different names all doing similar conversions. Would it be beneficial to combine these into a single function with keyword argument inputs so that we are just referencing this one function with different parameters as we need them? Just a thought I'm writing down in case this appeals to you, no need to do it here.

```python
def convert_time(my_time_in_whatever_frame, input_frame="TT", output_frame="ET"):
    # Some logic to select your input functions
    ....
    # Vectorize the spiceypy calls
    np.vectorize(func, ...)
```

_Originally posted by @greglucas in https://github.com/IMAP-Science-Operations-Center/imap_processing/pull/1295#pullrequestreview-2573330683_

vineetbansal, 2026-04-23:
So should we move any `imap_processing` code calling `TTJ2000_EPOCH` to fetch the latest kernel using our REST api, furnish them to spice, and then call our functions? (the code in `time.py` could do this fetching/furnishing itself).

vineetbansal, 2026-09-12:
Hi @tmplummer - I'm removing myself from this issue so someone else can take this on.