IMAP-Science-Operations-Center/imap_processing #1373: Calculate current s/c attitude from raw spin phase and angular momentum

greglucas, 2025-02-18:
### Summary

On I-ALiRT, we are getting packets back ahead of any historical/true attitude from SPICE. So we have to use either the predicted attitude kernels from SPICE or calculate it ourselves from the raw onboard data that comes down with the I-ALiRT packet. The onboard data should be more accurate in case the spacecraft drifts out of any nominal SPICE predictions, so we should use that if possible.

The onboard data that we have access to are:
* spin period estimate
* spin phase angle estimate (@ next 1 pps) NOTE: This appears like we might want to look at the historical previous packet to get the angle
* inertial RA of system angular momentum vector (0 -> 2pi, 16 bit unsigned int LSB of 9.5875e-5 rad/count)
* inertial dec of system angular momentum vector (-pi/2 -> pi/2, 16 bit unsigned int LSB of 4.7938e-5 rad/count)

Validity flags: 
* spin period validity (0: invalid, 1: valid)
* spin phase validity (0: invalid, 1: valid)
* attitude estimation source (0: star tracker, 1: sun sensor)
* catbed heater flag (0: off, 1: on)
* autonomy application validity flag (0: not operational, 1: operational)

The spin_phase is the angle rotating about the direction of the angular momentum vector. So the algorithm is something like: fix the RA/dec angular momentum vector in our reference coordinate system, then spin the spacecraft around with the proper spin phase angle after that. Use the geometry angle offset routines we already have for each instrument after that to calculate look direction of individual instruments.

### Requirements

  - [ ] Write up the math for calculating the look direction of each instrument from the raw packet data
  - [ ] Implement a lookup function similar to the geometry functions we already have for calculating instrument look direction
  - [ ] Verify that the predicted attitude kernels and this manual calculation match expectations
  - [ ] Extra credit:
    - [ ] Add checks for when to use raw data or not based on validity flags. Need to check with guidance and nav team about this. Maybe we just don't produce any I-ALiRT data when any of these flags are out of nominal...
    - [ ] Add additional warnings/checks when predicts don't match actual onboard calculations or things drift out of some threshold.

alastairtree, 2025-03-07:
This is great Greg, thanks for getting it fleshed out. Sounds good to me. This might just be elaborating what you have already spec'd, but for MAG at least I think we will need:
- to be able to calculate the MAG look direction based on our science acquisition timestamp rather than I-ALiRT packet time (our timestamp is just MET/SCLK value in seconds plus and a 16bit subsecond counter). Maxine already has code to go from our values to MET, but the point is that you want look direction at the science time and not the packet time. This means interpolation of look direction between several I-ALiRT packets and not just the look direction at the time of each packet. Mag data is spread over 4 seconds so you would need to be able to know the MAG look direction at any point in the last 4 seconds. 
- To handle multiple sensors - MAG has 2 sensors, so 2 instrument positions, 2 look directions, 2 science timestamps to use to make the calculations
- a function to transform a vector as measured while spinning into the spacecraft based co-ordinate system that is not spinning. A MAG vector is [x,y,z] at time t, and is based on the instrument measurement frame of reference - so the raw data is spinning. Once you have the MAG look direction in the spacecraft frame/co-ord system at the science measured time, you will need a function to be able to transform a MAG measured vector into your reference co-ord system. I think that is just a bit of matrix math on top of what you have to get the MAG look directions. 
If you do that transform for a few mins of mag packets (say) you have the vectors as 2 sets - as measured (spinning) and also in the space craft frame (despun). We need both sets for different calculations to remove noise.
- Once you have every the MAG vector in whatever well understood co-ords system you are using you need to be transform them into other frames of reference for the end products, so i think we are outputting the vectors to GSE, RTN, GSM. Not sure if other instruments have other frames as well, but I guess you are implementing a function to transform a vector to each of those frames?

All of the above are just a function call in spice, transforming a vector at a given time from one frame to another, but without spice, I think you need that vector transform function(s) in addition to your function to get the instrument look direction?

greglucas, 2025-03-10:
Thanks for the comments, @alastairtree!

> to be able to calculate the MAG look direction based on our science acquisition timestamp rather than I-ALiRT packet time (our timestamp is just MET/SCLK value in seconds plus and a 16bit subsecond [...]

To elaborate here, this ticket is purely for implementing the math to calculate the angles based on the spacecraft packet data. I assume each instrument will need to look at different packet's pointing information and do different things with this information. Like you say, you may only want the first packet's direction if that is when the measurement was taken. That is instrument specific though and each instrument can handle that how they desire.

> All of the above are just a function call in spice, transforming a vector at a given time from one frame to another, but without spice, I think you need that vector transform function(s) in addition to your function to get the instrument look direction?

I agree with this. My hope (I have not looked into this yet though) is that we could use SPICE for this, except rather than a query by time I'd go to the lower level transforms themselves saying: "Transform my vector from my_frame to X", where my_frame is a new frame we've calculated based on the s/c packet rather than getting that from SPICE kernels.

closed 2025-05-07 by laspsandoval