IMAP-Science-Operations-Center/imap_processing #1474: SPICE repoint - get mid time between repoints

subagonsouth, 2025-03-17:
### Algorithm Description:

This is for a common function that will get the mid-point time between the end of one repoint and start of the next. Input could be a MET time or maybe a repoint ID (number).

tech3371, 2025-08-20:
This is build from function that Sean has PR open, https://github.com/IMAP-Science-Operations-Center/imap_processing/pull/2061. This function just takes the np.mean(get_pointing_times()).

tech3371, 2025-08-20:
This works for the ENA because repoint times are filtered. pointing contains the repoint manuaver but ENA code filters the manuaver out.

Complication can come from what we define as pointing time is.
* repoint end to next repoint end (this is done in the pointing table but the code uses repoint files.)
* repoint end to next repoint start (This is what Sean did)

PR #2167 (2025-09-05): ENH: repoint mid-time utils function
# Change Summary
closes #1474 #1475

## Overview

Repoint mid-point utils functions.

## Updated Files

- imap_processing/spice/repoint.py
   - mid-time between repoints
   - repoint mid-time spacecraft state

## Testing

- imap_processing/tests/spice/test_repoint.py
closed 2025-09-05 by tech3371