IMAP-Science-Operations-Center/imap_processing #1489: Ultra l1b - test annotated data

laspsandoval, 2025-03-18:
### Summary

Nick Dutton generated some input/output data for annotated events code and we need to test our function. The name of his test data is:

input: IMAP-Ultra90_DirectEventsList_reduced.csv
output: annotated_events_reduced.csv

I have attached code to do this so it just needs to be cleaned up and it should be good.

### Requirements

The more specific the better.
  - [ ]
  - [ ]

laspsandoval, 2025-03-18:
Something like this for test_ultra_l1b_annotated.py:
@pytest.mark.external_kernel()
def test_get_particle_velocity_test_data(kernels):
    """Tests get_particle_velocity function with test data."""

    spiceypy.furnsh(kernels)
    test_path = imap_module_directory / "tests" / "ultra" / "test_data" / "l1" / "ultra-90_raw_event_data_shortened.csv"
    df = pd.read_csv(test_path)
    instrument_velocity = get_de_velocity((df["Xf"],df["Yf"]),
                                          (df["Xb"],df["Yb"]), df["d"],
                                          df["TOF"].values)

    et = spiceypy.str2et(df["Epoch"].values)

    frame_velocities = get_annotated_particle_velocity(et,
                                                       instrument_velocity,
                                                       SpiceFrame.IMAP_ULTRA_90,
                                                       SpiceFrame.IMAP_DPS,
                                                       SpiceFrame.IMAP_SPACECRAFT,)
    particle_velocity_spacecraft = frame_velocities[0]
    particle_velocity_dps_spacecraft = frame_velocities[1]
    particle_velocity_heliosphere = frame_velocities[2]

    v_mag_instrument = np.linalg.norm(instrument_velocity, axis=1)
    v_mag_spacecraft = np.linalg.norm(particle_velocity_spacecraft, axis=1)
    v_mag_dps_spacecraft = np.linalg.norm(particle_velocity_dps_spacecraft, axis=1)
    v_mag_heliosphere = np.linalg.norm(particle_velocity_heliosphere, axis=1)

    vhat_instrument = instrument_velocity / v_mag_instrument[:, np.newaxis]
    vhat_spacecraft = particle_velocity_spacecraft / v_mag_spacecraft[:, np.newaxis]
    vhat_dps_spacecraft = particle_velocity_dps_spacecraft / v_mag_dps_spacecraft[:, np.newaxis]
    vhat_heliosphere = particle_velocity_heliosphere / v_mag_heliosphere[:, np.newaxis]

    assert np.allclose(vhat_instrument[:, 0], -df["vhatX"].values, atol=1e-3)
    assert np.allclose(vhat_instrument[:, 1], -df["vhatY"].values, atol=1e-3)
    assert np.allclose(vhat_instrument[:, 2], -df["vhatZ"].values, atol=1e-3)

laspsandoval, 2025-03-18:
And something like this for ultra_l1b_extended.py:

def get_de_velocity(
    front_position: tuple[NDArray, NDArray],
    back_position: tuple[NDArray, NDArray],
    d: np.ndarray,
    tof: np.ndarray,
) -> NDArray:
    """
    Determine the direct event velocity.

    Parameters
    ----------
    front_position : tuple
        Front position (xf,yf) (hundredths of a millimeter).
    back_position : tuple
        Back position (xb,yb) (hundredths of a millimeter).
    d : np.array
        Distance from slit to foil (hundredths of a millimeter).
    tof : np.array
        Time of flight (tenths of a nanosecond).

    Returns
    -------
    velocities : np.ndarray
        N x 3 array of velocity components (vx, vy, vz) in km/s.
    """
    tof_seconds = tof * 10**-10  # convert to seconds

    if tof[tof < 0].any():
        logger.info("Negative tof values found.")

    # distances in mm
    r = np.empty((len(d), 3), dtype=np.float32)
    r[:, 0] = (front_position[0] - back_position[0]) * 0.01
    r[:, 1] = (front_position[1] - back_position[1]) * 0.01
    r[:, 2] = d * 0.01

    v = r / tof_seconds[:, np.newaxis] * 10**-6  # mm/s -> km/s

    return v

greglucas, 2025-10-21:
I _think_ this is now complete as well?