IMAP-Science-Operations-Center/imap_processing #1746: Maps - Use iterative standard deviation to compute obs_date_range

subagonsouth, 2025-05-16:
I think that Jon sent us a link to a way we can do this in his email titled "iterative standard deviation". 

[StackExchange thread](https://math.stackexchange.com/questions/2148877/iterative-calculation-of-mean-and-standard-deviation/2148949#2148949)

Looks like we can project 3 values separately: `all ones, obs_date (per-pixel), obs_date**2` and then compute $\sigma_{obs\_date}$

_Originally posted by @nkerman in https://github.com/IMAP-Science-Operations-Center/imap_processing/pull/1743#discussion_r2093426871_

nkerman, 2025-05-28:
I'm starting to look into this on branch: `nkerman/std_obs_date` - but, while I have been able to implement the calculation, it seems _really_ high. I was using 4 different PSETs, and as I decreased the difference in epoch between them, the `obs_date_range` stayed higher than the simple difference between max_epoch - min_epoch.

nkerman, 2025-05-28:
Nevermind! I think I was missing a squared term somewhere; I've got it looking good now!

closed 2025-05-29 by nkerman