IMAP-Science-Operations-Center/imap_processing #2111: MAG l1d l2 - Use correct instrument frame for geometry

subagonsouth, 2025-08-27:
### Algorithm Description:

A new version of the instrument frame kernel changed from having just one MAG frame to having 3. IMAP_MAG_BOOM, IMAP_MAG_I, and IMAP_MAG_O. The correct instrument frame needs to be used to transform MAG data into the s/c reference frames.

From slack convo with @maxinelasp about this:

> Tim Plummer
> At what level of MAG processing do you do that transform?
> 
> Maxine Hartnett
> Just in L1D and L2. I think it will be a simple change, if I set the frame specifically when creating the class instead of defaulting to IMAP_MAG, we can specify MAGO/MAGI at that point and then everything should just work
> 
> Tim Plummer
> Do you think that this will impact any validation for those products?
> 
> Maxine Hartnett
> Yes, probably
> L1D validation has data but isn’t complete. L2 validation is done and did use the SPICE kernels. Hopefully it’s as easy as asking the MAG team to re-generate the validation data, since they used the SPICE kernels we had.

maxinelasp, 2025-09-03:
The MAG team has requested that this be completed before launch with the "idealized" kernel for validation purposes. They would also like the non-idealized real kernel before launch if possible, but it isn't needed for validation, only to generate calibration files.

subagonsouth, 2025-09-05:
Part 1 has been completed in PR: #2177